sinara-hw/Driver #1: [RFC] Driver: current driver for laser diodes

jordens, 2019-12-29:
We may be looking at designing a diode laser current driver in the future. Comments and secondments welcome.

### General notes

* Low-noise current controller for (low-power) laser diodes.
* Form factor TBD. May end up being a Stabilizer mezzanine or a separate board that interfaces with Stabilizer or a separate EEM
* Either two versions for 200-250mA and 1.5-2A range or variants or one design that fits both
* Different from the [coil current driver](https://github.com/sinara-hw/meta/issues/23) in that it has smaller range, drives diode-like loads, doesn't need that much stability, but needs modulation features.
* Linkage to Thermostat: ethernet, Stabilizer: analog, and Zapper: analog
* Specifications and performance goals similar to the open designs:
  * [Libbrecht-Hall](https://aip.scitation.org/doi/10.1063/1.1143949)
  * [Libbrecht-Hall-Durfree](https://arxiv.org/pdf/0805.0015)
  * [Libbrecht-Hall-Seck](https://arxiv.org/pdf/1604.00374)
  * [JQI](https://github.com/jqiamo/current-controller/tree/v2-updates) to be consolidated, based on Libbrecht-Hall-Durfree
* Compare to proprietary things:
  * [Vescent](https://www.vescent.com/products/electronics/d2-105-laser-controller/)
  * [ILX Lightwave](https://www.newport.com/f/precision-laser-diode-drivers-ldx-3200)
  * [Toptica](https://www.toptica.com/products/tunable-diode-lasers/laser-driving-electronics/)
  * [Koheron](https://www.koheron.com/): uses LT3045, LT3261, OPA567, LT6655
  * [MOGLabs](http://www.moglabs.com/diode-laser-controller.html)
  * [SISYPH SMC11](http://www.sisyph.com/smc11-puy-mary-ultra-low-noise-current-source)

### Variant A

  * 250 mA
  * 2V compliance
  * single quadrant
  * assumptions/requirements w.r.t cathode grounded, anode grounded TBD
  * 0.5 µA resolution, digitally controlled (~100 kHz resolution, 200MHz/mA tuning)
  * 10 kHz small signal bandwidth
  * RMS noise 10Hz-10MHz: 0.3 µA
  * RMS noise 10Hz-10kHz: 0.05 µA
  * Stability (<10Hz) TBD
  * Linearity/absolute accuracy: 2% TBC
  * tempco 5 µA/K (1 MHz/K, 20 ppm/K) TBC
  * high demands on laser diode safety guarantees: no turn on/off transients, no chance of ESD/ground potential difference getting through to the diode, benign behavior and recovery from short circuits and open connections
  * could be some coarse+fine additive or multiplicative hybrid solution to achieve resolution TBD
  * has a modulation/error signal input that is both sampled by an ADC (at 100 kHz) and a high frequency bypass (10 kHz to 100 MHz) that goes onto a bias tee (together with the low frequency current output from the controller) on the output to the laser. Or alternative approaches TBD.

### Variant B (otherwise same as A)
  * 1.5A
  * 3 V compliance
  * bandwidth and resolution much less critical
  * noise requirement at a similar relative level as A

jordens, 2020-01-14:
Added the [AOSense](https://aosense.com/solutions/integrated-laser-controller/) module to the top post. Thanks Mariusz for pointing it out.

dtcallcock, 2020-03-01:
Is there any reason to think Variant B wouldn't also make a nice current supply for the few-Gauss field coils found on most ion traps (or at least a good starting point for one - a factor of 2 more current might be nice)?  Is it funded or was that just the A Variant?

jordens, 2020-03-01:
You'd have to deal with the inductive load. Both variants are funded.
But for shim coils it looks like overkill: around 10 A-turns you can dimension the coil to be in the range of a simple single-opamp voltage-to-current converter running of Zotino/Fastino since there are no grounding restrictions. This would also be bipolar while Driver will not necessarily be.

fimek, 2021-02-08:
Right now there are following problems regarding laser driver:
- combining the expected noise level with reasonable power consumption,
- laser driver precision (meeting the 16-bit DAC precision and resolution),
- bipolar transistors instead of unipolar transistors (due to their more promising noise parameter),
- DAC noise density within 100 kHz bandwidth (seems to be the most challenging - the DAC cannot be BW limited for noise reduction).

It should be decided whether it is sensible to test just the unidirectional current source, or the current source sould be designed as bidirectional (compatible with both common anode and common cathode lasers).

fimek, 2021-02-08:
My strategy briefly explained:
1. DAC: non buffered DAC, where the only output noise is the one generated by the R-2R ladder. Best achievable parameters: 7...9 nV/sqrt(Hz). Proposed ICs: DACx1001, AD5791 (seems similar in performance).
2. DAC output voltage swing: maximum available: 0...10V. If the reference voltage for the current source is too high - can be divided leading to DAC output noise reduction. 0...10V will match the current source optimum driving voltage.....
3. ....which depends on the output transistor emiter resistance. 40 Ohms seems OK, together with ref. voltage of up to 10 V result in 250 mA of maximum laser current. 
4. Current source has to be built around the opamp with En <= 3nV/sqrt(Hz), low input current noise (together with DAC ladder resistance would degrade noise performance) <= 0.5 pA/sqrt(Hz), low bias current and appropriate  input voltage offset. Fortunately - such opamps exist: AD8675.
5. By using the additional JFET we can get rid of output transistor base current and maintain DC stability.
6. Power supply: +15V /- 5V (can be +15V/-15 V either).
7. It's a bit tricky to generate a current source with grounded anode/cathode, because the reference voltage for the current source must be referred to Vdd or Vss. I'd rather avoid differential amplifiers (due to remarkable additional noise contribution). I' would use LTC1043 (or similar) instead (the BW is sufficient). 
[image]