sinara-hw/Driver #3: Driver module schematics review

akaminska, 2022-04-04:
Please find attached the preliminary Driver schematics for review. Please note:

- This project is focused on the low-noise current source. The higher power current source is still under design and will be added later.
- The laser protection subsystem ("protection", schematics on the last page of the attached pdf) will be updated, please omit it for now in your review.
[driver_3_en_16032022.pdf](https://github.com/sinara-hw/Driver/files/8410830/driver_3_en_16032022.pdf)

filipswit, 2022-05-18:
PDF version of schematics:
[AI-ARTIQ Driver v0.2.pdf](https://github.com/sinara-hw/Driver/files/8715866/AI-ARTIQ.Driver.v0.2.pdf)

nkrackow, 2022-05-20:
I might have found an issue:
The `TEMP` node is connected to `T11` base for both driver output protection circuits, right? Let's say there is no over-temperature event, so the temperature sensor open drain output can be ignored. Now if the `IC27` flipflop is set for one side, it will drive both `T11` transistors in both output stages in parallel. Or am I missing something?

I know the protection circuit will go to the headboard but the same applies there.

nkrackow, 2022-05-20:
I also did some lengthy staring at the floating DAC ground setup again and while it's still very confusing I convinced myself that this might work..

Also the rest of the relay logic looks good I think. Just that we need to load new outputs into the shift register twice to change the state of K0 with the flipflop. Can't we use a d-latch eg. [this](https://www.ti.com/lit/ds/symlink/sn74lvc1g373.pdf?ts=1652989204874&ref_url=https%253A%252F%252Fwww.google.com%252F)? That would also save all of the nand gates since we can just use the inverted state of the node that now goes from K1 to the first nand gate as the enable signal of the latch.

Last thing is that I remember @jordens wanted to maybe sample the voltage over the output current shunt resistor (the one that gives feedback to the opamp). Is that still relevant?

jordens, 2022-05-21:
* The circuitry on protection.SchDoc and I2C_EXPANDER should be on the header board.
* Can we unify CS_OUT_MON and V_LASER_MON? They both appear to have the same functionality (measuring voltage across the load). I don't see the point in having two. Maybe RL2B to switch the measurement channel (Pin 3) between R130 (Pin 2) and the laser (VLASER J2 Pin 1).
* It would be reaslly desirable to get laser diode current monitoring. Maybe we can rejigger the relay circuitry on the header board to achieve that (use a ~1 Ohm shunt for R130 and put that always in series with the sink/source, independent of whether the laser diode is shorted/bypassed). Or other ideas.
* @nkrackow please also thoroughly review the SPI bus usage and assignment w.r.t. how their STM32 peripherals would be used and whether there are interdependencies and constraints.
* The HP source/sink doesn't need to dissipate that much power. It doesn't need to be so low noise. A 0.1 R or even lower shunt should be fine. Also lower supply voltage (to reduce the drop across the power transistor).
* Can we also add NTCs and some simple circuitry to monitor the heatsink temperature using the CPU ADCs?
* How fast is the current sink/source (including all the output circuitry) to the laser diode.
* Can we make use of Stabilizer's ADCs/DACs (J6)? I think one ADC is used for the LF error signal sampling (is that done via SMA separately from the header board?) But there is one more ADC and two DACs. If the current source isn't very fast it may make sense to have at least a way to couple the DACs AC onto the sink/source currents with C jumpers. Just as a DNP option.

jordens, 2022-05-24:
@schiemangk  @j-hirsch you also wanted to have a look.

gkasprow, 2022-05-31:
I moved the protection schematics to the laser module.
I also used an 8k HDMI cable which has individually shielded pairs to transfer all signals from the laser driver to the laser module.
Here are the schematics:
[Laser module PCB.PDF](https://github.com/sinara-hw/Driver/files/8807392/Laser.module.PCB.PDF)
[Frequency REF module PCB.PDF](https://github.com/sinara-hw/Driver/files/8807396/Frequency.REF.module.PCB.PDF)
[Micro_DSUB_IDC.PDF](https://github.com/sinara-hw/Driver/files/8807407/Micro_DSUB_IDC.PDF)

bbradzik, 2022-06-01:
> I might have found an issue: The `TEMP` node is connected to `T11` base for both driver output protection circuits, right? Let's say there is no over-temperature event, so the temperature sensor [...]

Good point. We will add components to isolate both (LN and HP) TEMP nodes.

bbradzik, 2022-06-01:
> Also the rest of the relay logic looks good I think. Just that we need to load new outputs into the shift register twice to change the state of K0 with the flipflop. Can't we use a d-latch eg. [...]

About month ago I draw similar circuit with level triggered D flip flop. Greg has told me to swap it with edge triggered D flip flop commenting (as far as i recall) that it will be more proof against unwanted enabling of the laser.

> Last thing is that I remember @jordens wanted to maybe sample the voltage over the output current shunt resistor (the one that gives feedback to the opamp). Is that still relevant?

The idea is good, however it would require to add two instrumentation amplifiers.
Would four voltage measuring points (two per laser diode) be still relevant then?

nkrackow, 2022-06-01:
> About month ago I draw similar circuit with level triggered D flip flop. Greg has told me to swap it with edge triggered D flip flop commenting (as far as i recall) that it will be more proof against unwanted enabling of the laser.

I see, then we'll just do some more flipping and flopping with the CPU via I2C. :)

> The idea is good, however it would require to add two instrumentation amplifiers.
Would four voltage measuring points (two per laser diode) be still relevant then?

Yeah I think this would be on the wishlist from QUARTIQ side. As stated in my response to Roberts points, we could just have the four signals going back to the Stabilizer MCU ADCs (`JADCx_INx_x` on the Stabilizer header, we don't need differential there) via the 8K HDMI cable. I think there should be some lanes available still @gkasprow?

nkrackow, 2022-07-13:
Hi, one more thing that just came up on our side is that the software needs to detect if Driver is present on Stabilizer. A nice way to do this would be to read an EEPROM via the I2C line that goes to Driver anyways. Would be great if we could still tag that onto the prototype.

nkrackow, 2022-07-20:
Ok, one last last thing is that it could be nice to have the global overtemperature node `TEMP` also available on an MCU GPIO if we can fit it somehow.
Sorry, I'll stop adding things here as soon as someone tells me Driver is finalized. ;)

closed 2022-09-22 by filipswit