sinara-hw/Driver #5: PDF version

nkrackow, 2022-08-10:
Hi, could we also get a PDF version of the schematic+layout if that's not too much of a hassle?

filipswit, 2022-08-10:
Sorry, I forgot about PDFs
Schematics are in sink variant just to know which other components will be mounted
[AI-ARTIQ Driver v1.0_sch LNsiHPsi.pdf](https://github.com/sinara-hw/Driver/files/9299079/AI-ARTIQ.Driver.v1.0_sch.LNsiHPsi.pdf)
[AI-ARTIQ Driver v1.0_pcb print.pdf](https://github.com/sinara-hw/Driver/files/9299018/AI-ARTIQ.Driver.v1.0_pcb.print.pdf)
[AI-ARTIQ Driver v1.0_3D pcb print.pdf](https://github.com/sinara-hw/Driver/files/9299019/AI-ARTIQ.Driver.v1.0_3D.pcb.print.pdf)

filipswit, 2022-09-19:
[AI-ARTIQ Driver v1.0rc1_sch LNsiHPsi.pdf](https://github.com/sinara-hw/Driver/files/9596875/AI-ARTIQ.Driver.v1.0rc1_sch.LNsiHPsi.pdf)

filipswit, 2022-09-22:
Actual PDFs are in release.

closed 2022-09-22 by filipswit