sinara-hw/Driver #6: Driver schematic/layout review

nkrackow, 2022-08-10:
We've had a good look at the layout now and it looks very nice! We're of course no experts in layout so we didn't check for clearance or trace thickness or things like that. 
Just a few comments about the layout: 
- Did you check the assembly with Stabilizer? Aka that there are no tall components potentially colliding with Stabilizers components?
- Are the right nodes on the HDMI pins that make connection first? The TEMP node should make connection first when plugging in and break last when pulling out. 

There are two points from the schematic review still:
- https://github.com/sinara-hw/Driver/issues/3#issuecomment-1190233600 We would really like to also have control over the TEMP node from the CPU. The best would be if the TEMP node is usually pulled low (aka overwriting the relay logic and forcing the laser diode to be shorted) until a CPU pin goes high. 
- https://github.com/sinara-hw/Driver/issues/3#issuecomment-1143563047 We would like to measure the voltage across the output shunt resistor to continuously be able to measure the output current.

jordens, 2022-08-10:
> The TEMP node should make connection first when plugging in and break last when pulling out.

The other way round.

filipswit, 2022-08-12:
> https://github.com/sinara-hw/Driver/issues/3#issuecomment-1190233600 We would really like to also have control over the TEMP node from the CPU. The best would be if the TEMP node is usually pulled [...]

So on Driver module, I could use `LPTIM2_OUT`, it's not used so far. Is it fine to just short `TEMP `and `LPTIM2_OUT`?

nkrackow, 2022-08-15:
We would suggest the following solution: 

Since we need some kind of decoupling from the transistor bases anyways (https://github.com/sinara-hw/Laser_Module/issues/1), `TEMP` becomes a logic signal that is valid either high or low. 
Then we can have a big pulldown resistor (~10K) on `TEMP` the header board that usually makes sure that `K1` cannot be enabled (maybe we don't even need that with npn transistors since nothing is actively driving the node). 
The MCU GPIO pin is then connected to `TEMP` via a small resistor (~100 ohm / whatever fits with the base current limiting resistors) and can therefore drive `TEMP` high, making it possible to activate `K1` via the other relay logic.
The LM75 can then still force `TEMP` low. 

We could even call this an interlock at this point, but we're fine with naming it `TEMP`, too.

filipswit, 2022-09-19:
The final version is already on repo. I've splitted TEMP and MCU_Interlock, so we have two signals which should be compared on Laser_Module board. I've places them on pins 13 and 19, We can't use  1st pin because it's wired as diff pair so we'd lose it.

jordens, 2022-09-20:
@filipswit Great work. Thanks for the release! Let's hope and push that this gets manufactured in time.
Could you check the issues and either close them when there is nothing to be done or ping the respective people if unresolved and waiting for input?

gkasprow, 2022-09-20:
@filipswit please update the connector as we discussed. I implemented changes on my side
[image]

filipswit, 2022-09-21:
Updated pinout in https://github.com/sinara-hw/Driver/commit/d203db5ae0c88f6e30ad801b221f2400497f251d

closed 2022-09-21 by filipswit