sinara-hw/Driver #10: (inconclusive) LN DAC reference voltage not stable

nkrackow, 2022-12-09:
The reference voltages of the DAC on the low noise (LN) side seem to be off under some circumstances. Note that all voltages are with respect to DACGND.
If I just power up Driver and don't connect it's output I see +4.1,-5.0 V. The DAC output is also not at the correct value of -5 V, but at around -0.9 V. 
If I short the output, I actually see the expected 250 mA (0x00 DAC reset code) on the output and the reference voltages seem to come closer to nominal +- 5 V (actual value: ~4.95, -4.98). The DAC output is now as expected at -5 V aka at the negative reference.
If I set close to 0 output current (max DAC code), I also see relatively good references (~4.95, -4.98) and the expected DAC output of ~4.95 V. The output current is close to 0.

However in the middle, when the DAC outputs ~0 V, the positive reference seems to be off again, showing about 4.8 V. 
Note: The HP side reference voltages (with or without R77 shorted) are always very close to ideal at ~ +4.995,-4.995 V. 
I didn't yet understand why the reference voltage is off in the middle of the DAC output range and better (also not ideal) at the extremes. Maybe a grounding issue with the DACGND? We couldn't see a problem with it right away but we very well might have missed something.

nkrackow, 2022-12-14:
The HP side references are rock solid under all possible combinations of the following:
- open output
- shorted output
- high output set current
- near-zero output set current
- some other set current

bbradzik, 2022-12-14:
I'm working on communicating with DAC over SPI the crude way (Arduino) to be able to set different codes and reproduce Your problem. I should have some info in few days.

bbradzik, 2022-12-21:
OK, i have SPI working.
For the LN driver I set three values of current: zero, half scale, full scale. Each setting stays for 3s and cycle. 
Without load (loaded with 1 ohm) i get RefP5 of 4.999V RefN5 of -4.998V maybe changing by some 1mV while changing codes. DAC output also is close to -5V; 0V; +5V as expected. 
Without load (output open) i get RefP5 of 4.999V RefN5 of -4.998V but DAC output is now changing: +4.98V: -0.43V; +2.04V plus IC15B is hot. So probably input diodes of the opamp are conducting and loading the ouptut of the DAC. Nevertheless open output is not a normal condition for a current source.

[image]

*All voltages measured wrt DACGND.

nkrackow, 2022-12-26:
Yes, I have also noticed that the opamp gets hot when the output is open. Even though it's not the standard operating condition and seems to not brak anything I'd say that's not optimal. 

Summing up our voltage readings: 
- We see the same voltages/conditions with an open output.
- Wit a loaded output (I simply shorted it) there is a slight difference: I still see slight shifts in the DAC reference voltages of a few 10s of mV while yours is stable to within 1 mV.

I'll check tomorrow how my reference looks like when I connect a small resistor.

nkrackow, 2022-12-28:
I have checked again with two diodes (0.9 - 1.4V drop depending on current) across the output now and this didn't improve things for me. The negative reference is still rock solid but the positive is at:
- 4.935V for close to 0 A aka DAC output close to positive ref.
- 4.830V for 100 mA current aka close to 0V DAC output aka midpoint between pos. and neg. ref. 
- 4.875V for 200 mA current aka ~ -3V DAC output