sinara-hw/Driver #11: (inconclusive) no HP output current

nkrackow, 2022-12-09:
After shorting R77 according to https://github.com/sinara-hw/Driver/issues/8 I see good rails for the HP output stage.
The DAC also seems to behave as expected under all observed circumstances (output shorted/open, different DAC output setpoints). Unfortunately I never see any current on the output, no matter what current I set. After the follower-divider stage (the purpose of which is not entirely clear to me) I see a small voltage wrt. the positive 9 V rail, but the output opamp (IC15A) is not happy, aka there is a voltage drop across its inputs. 
I didn't get any further so far.

bbradzik, 2022-12-14:
_> the follower-divider stage (the purpose of which is not entirely clear to me)_
The important issue with higher current linear current sources is power dissipation. Especially in the sensing resistor. The power is dissipated also in the power transistor and the load but the sensing resistor is a component which determines how stable the output current will be. That is why we don't want to dissipate a lot of heat in it. Considering P=I^2*R we want the resistance to be small. So the voltage across the resistor  will be low. Wee need to create this low voltage for the current source input. This is what the divider is used. We can't just use out 10V from the DAC. We also don't want to load the DAC's ouptut resistive ladder with the resistive divider because we will loose linearity. So the follower is used to prevent DAC's output from being loaded.

jordens, 2022-12-14:
How does this work out quantitatively? The transconductance of the output stage (including its opamp) is 1/R (the resistance of the shunt). The shunt is 0.68 Ohm, the follower has a gain of about 1/40. Hence the overall transconducance is 0.36 A/V and for 10 V max DAC output max output current is 3.6 A. Is that correct?

bbradzik, 2022-12-14:
Very close.
10V*1/41=0.244V
Across 0.068 ohm resistor this gives 3.58A max.

However there is inconsistency: 0.68R resistor value is printed on the schematics while ordered and soldered is 0.067R.
Now the problem with 0.068R is that the lead/wire/contact resistance is comparable.

The initial components were 0.68R and five 10k resistors in the divider (2V full scale output) but at some point it was changed and overlooked.

jordens, 2022-12-14:
Ah. Thanks for clarifying. Let's update schematics then.

bbradzik, 2022-12-14:
Unfortunately some more issues are in the HP driver.
1. both drivers start with the full current after powering up, creating high inrush current. This off course is mostly due to HP driver
2. the power transistor of the HP driver works too close to the envelope of the Safe Operating Area. The effect is more serious when the driver's output is shorted to ground. After connecting laser diode it is getting better. There are two solutions to that: use bigger transistors  (eg. TIP125) or re-calculate the divider so it gives 2.5A max.

SOA chart for presently used transistor is below:
[image]

jordens, 2022-12-14:
For AI-ARTIQ this isn't a problem yet, as current hasn't been more than 1A.
Stability is also not such a big issue: in steady-state the resistor heating will only lead to a departure from linearity.

nkrackow, 2022-12-14:
The problem was simply that the base and emitter connections of the output transistor were not connected. The leads were just free in the terminals with the screws wide open. Only the collector lead was tightened. Same was true for the LN side but there it happened to make contact without tightening for me. This was the state we received the boards in.

nkrackow, 2022-12-14:
> 10V*1/41=0.244V
Across 0.068 ohm resistor this gives 3.58A max.

The actual current is relatively consistent with this, just about 5% low. This makes sense when taking parasitic resistances of the terminals etc. into account. Those would be an extra 3.5 mΩ then.

bbradzik, 2022-12-14:
> The actual current is relatively consistent with this, just about 5% low. This makes sense when taking parasitic resistances of the terminals etc. into account. Those would be an extra 3.5 mΩ then.
Seems legit.

I'm glad that You found the error. It would be difficult to observe same behaviour on my board.

nkrackow, 2023-06-29:
closing since this was a specific issue with the first Driver

closed 2023-06-29 by nkrackow