sinara-hw/Driver #12: found PCB bugs and errors

bbradzik, 2022-12-21:
1. swapped U_LN and U_HP text on top layer
[image]

bbradzik, 2022-12-21:
T14A (top) and T14B (bottom) have swapped D and S pins. The error is on the schematics and PCB. A solution is to desolder both transistors (with ESD precautions), flip it and solder on the back (pin 1 stays where it was, pin 2 should be connected to pin 3 of the footprint )

[image]

[image]

nkrackow, 2022-12-28:
Swapping the pins worked nicely for me. The LN side gives relatively accurate current readings now. Unfortunately the scale factor for the HP side is rather impractical and errors in resistance and other factors lead to significant inaccuracy. I'll open a new issue for this.