sinara-hw/Driver #13: High Power current sense

nkrackow, 2022-12-29:
The high power side current sense setup is not very accurate because of the combination of the very small current sense resistor (0.068 Ω) and the small translation scale in the opamp. (Using [this](https://github.com/sinara-hw/Driver/issues/12#issuecomment-1362180968) fix)
[image]
Only a very small part of the ADC range is utilized and the readings are pretty noisy .There is also an offset of about 0.2 A.
The voltage readings of both sides are also somewhat noisy. The following gives a rough idea of the offsets/noise for voltage and current readings:
[image]
There are current jumps from 0 to 0.2/2.0 A for LN/HP outputs on the left/right over a diode.

bbradzik, 2023-01-04:
Here is a simplified schematic of the present situation in the HP part and the proposed modifications. The four wire Kelvin sensing won't be needed in proposed solution. Please let me know what You think and I will ask Ania to organize parts for me to test it and for You to modify Yours.
[image]

nkrackow, 2023-02-13:
Hi @bbradzik, sorry for the late reply.
The changes to the sense part look sensible to me. Looking at the current setup it seems like we could also just change the 10K resistor to ~200ohm for now to make better use of the ADC range?
I don't know what the changes in rail and opamp offset voltages entail, so I can't comment on that.

jordens, 2023-02-24:
Would INA194 be a more robust accurate solution for the high side current sense?