sinara-hw/Driver #15: Noise/ripple on the Stabilizer MCU ADC inputs

nkrackow, 2023-02-16:
I was wondering why the readings on the current and voltage readbacks are relatively noisy:
[image]
This is clear for the HP current channel #13, but the voltage readings should be using quite some dynamic range of the ADC. 
I first suspected the software/ADC setup procedure since we had problems with this in the past https://github.com/quartiq/stabilizer/issues/571 https://github.com/stm32-rs/stm32h7xx-hal/issues/374

However, I am now quite certain this is actual electrical noise on the nodes.
The samples of the ADC look like this and have about 1000 LSB PP (~1.5% fullscale) and 240 LSB RMS noise.
[image]

I made a small 50:50 resistor divider between GND and 3V3 directly on the Stabilizer ADC header with the 1.65 connected to an ADC input. Now I see much less noise of ~100 LSB.
[image]

I cant measure the noise on a multimeter but I can see significant ripple on the scope. Ofc noise is difficult to measure accurately and I just used a standard probe with an alligator GND lead (loop). This is directly on the connector pin that goes to stabilizer:
[image]
Frequency of the ripple is about 50 kHz. (sorry for the missing x axis)

The LN current measurements look much cleaner too:
[image]
45 LSB RMS and the pin looks much cleaner on the scope too:
[image]

Sampling was at 200 Hz with sampling time for one sample 2 ADC cycle times. This means most of the time the ADC was not sampling anything and the noise bandwidth is much higher than the signal bandwidth. Ofc. the 50 kHz ripple are not directly visible on the ADC like this.

The ADC sampling time (in the standard driver code at 810 ADC cycles) has an effect on the noise to about 64 cycles. Longer sampling times don't bring any benefit, at least not with the amount of electrical noise I was seeing anyways.

nkrackow, 2023-02-16:
The problem is that the ADC samples very fast (it could run at 100 kHz of faster) but does so only at a rate of 200 Hz. So I'm not sure what exactly to make of the spectrum: 
[image]
This is the PSD of the data with 1000 LSB PP (~1.5% fullscale) and 240 LSB RMS noise.

nkrackow, 2023-02-16:
Getting the ADC to sample much faster would require quite substantial changes to the software right now..

gkasprow, 2023-02-16:
Can't you just read the ADC as fast as possible filling entire RAM and then dump them to UART?

nkrackow, 2023-02-16:
Well yes but I'd have to write quite some code for that ;)

nkrackow, 2023-02-20:
Could you guys at Creotech (I think @bbradzik has the other Driver right?) check if you see the 50 kHz ripple on the voltage sense nodes? `CS_OUT_MON_LN` and `CS_OUT_MON_HP` on the schematic.

nkrackow, 2023-02-22:
> What amplitude to expect? Under what load and DAC code does it appear?

bbradzik, 2023-02-23:
The P1V5REF voltage is oscillating due to wrong output capacitor selection (red dot shows present situation): 

[image]

Adding 10uF/6.3V X7R MLCC capacitor across C89 on top layer kills the 50kHz oscillations.
However there is still one problem: the P1V5REF rail is still not clean because of the ripple coming from P12V0 power line.
I will prepare the simplest solution to that.

nkrackow, 2023-02-24:
Cool, yeah the voltage readings look a lot better now (jump from 0 to 100 mA over two diodes)
[image]