sinara-hw/Driver #20: Heatsink mounting

nkrackow, 2023-05-02:
The current heatsink solution is not really ideal yet for mounting. The PCB spacers take a lot of force in the horizontal direction and the fins of the heatsink inhibit correct hole placement. Note that the spacers in the back are not connected to the PCB.
[image]

gkasprow, 2023-05-04:
@jmoskwa did you design this heatsink in CAD or just took a piece of raw extrusion and simply drilled the holes?

filipswit, 2023-05-04:
It's Bartek's heatsink for RnD purpose, since we still don't have final one.

jmoskwa, 2023-05-04:
Yes, I designed it in CAD in the first place. The vertical distance between mounting holes was from th begening 30mm.
Here's the screen from the 3D model. On top the heatsing, beneath - aluminum plate, that was attached to th PCB.
[image]

gkasprow, 2023-05-09:
Did sb calculate the required thermal resistance?